amccaugh/phidl #56: latest version not working with gdspy v1.4

petebarry1988, 2019-05-20:
After downloading the latest version of phidl from pip (1.0.2), and the latest version of gdspy (1.4) I receive the following error when initialising a device (using the tutorial example). 

``` D = Device('MultiWaveguide')

AttributeError                            Traceback (most recent call last)
<ipython-input-6-ba05aab005b6> in <module>()
----> 1 D = Device('MultiWaveguide')

lib/python2.7/site-packages/phidl/device_layout.pyc in __init__(self, *args, **kwargs)
    445         self._internal_name = _internal_name
    446         gds_name = '%s%06d' % (self._internal_name[:20], self.uid) # Write name e.g. 'Unnamed000005'
--> 447         super(Device, self).__init__(name = gds_name, exclude_from_current=True)
    448         Device._next_uid += 1
    449 

lib/python2.7/site-packages/gdspy/__init__.pyc in __init__(self, name, exclude_from_current)
   4839     def __init__(self, name, exclude_from_current=False):
   4840         self.name = name
-> 4841         self.polygons = []
   4842         self.paths = []
   4843         self.labels = []

AttributeError: can't set attribute

amccaugh, 2019-05-20:
Thanks for the bug report--you're correct, gdspy 1.4 changed a few internal representations that phidl talks to, so I need to match those changes in phidl.

I'll have a new version out in a couple days though which will work with 1.4, but until then it'd be best to install the previous version of gdspy (e.g. `pip install gdspy==1.3.2`)

petebarry1988, 2019-05-20:
Yup, works for me. Thanks!

amccaugh, 2019-05-23:
Alright, I fixed everything and wrote some new tests to verify it's working 100%, but please let me know if you run into any bugs.  Thanks again for the bug report

closed 2019-05-24 by amccaugh