amccaugh/phidl #61: how to move device with labels?

joamatab, 2020-01-09:
Hi Adam,

we still use phidl 1.0.3 because one of our tests broke on the following releases

I just wanted to check with you whether this is a bug ^^

I find it convenient to be able to move a Device that has labels on it

Thank you for your work on Phidl, it's great!

```
from phidl import device_layout as pd
import phidl.geometry as pg
from phidl.device_layout import Device
from phidl import quickplot as qp

def test_label_move():
    """ test that when we move a device its label also moves """
    c = Device()
    E = pg.ellipse(radii=(10, 5), layer=1)
    label = pd.gdspy.Label(
        text="demo", position=(0, 0), anchor="o", layer=66, texttype=0,
    )
    c.add(label)
    c.add_ref(E)
    c.movex(10)
    print(c.references)
    assert c.references[0].origin[0] == 10
    assert c.labels[0].position[0] == 10
    qp(c)

if __name__ == "__main__":
    test_label_move()

```

amccaugh, 2020-01-09:
Hi @joamatab , thanks for the report -- this actually got fixed in 1.2.0!  (this specific commit:  https://github.com/amccaugh/phidl/commit/ad78ca1a9d0edccdd42ce4dbbfe09147f61ef69b ).  However, you need to use the phidl Device.label() function instead of manually adding a gdspy label., because gdspy labels don't have a .move() method.  

For instance, if I just change the :
```
    label = pd.gdspy.Label(
        text="demo", position=(0, 0), anchor="o", layer=66, texttype=0,
    )
```
line to 

`label = c.label( text = 'demo', position = (0,0), layer = [66,0])`

It looks like it works.  Note that texttype is treated like datatype is in phidl layers -- that is, if you want something with a (layer 66, textttype 0), you pass the `layer` argument (66,0) just like you would for a polygon.

```
from phidl import device_layout as pd
import phidl.geometry as pg
from phidl.device_layout import Device
from phidl import quickplot as qp

def test_label_move():
    """ test that when we move a device its label also moves """
    c = Device()
    E = pg.ellipse(radii=(10, 5), layer=1)
    label = c.label( text = 'demo', position = (0,0), layer = [66,0])
    c.add_ref(E)
    c.movex(10)
    print(c.references)
    assert c.references[0].origin[0] == 10

    assert c.labels[0].position[0] == 10
    qp(c)

if __name__ == "__main__":
    test_label_move()
```

joamatab, 2020-01-10:
the phidl label worked great!

thank you Adam!

closed 2020-01-10 by joamatab