amccaugh/phidl #67: pg.basic_die does not work for non square dies

jonnyfountain, 2020-03-12:
When I try to define a non square die using the pg.basic_die, the y-position and dimensions of marks is not scaled correctly.
[image]

amccaugh, 2020-03-12:
Good catch, thanks @jonnyfountain .  This will be fixed in the next version, but until then you can use the fixed code below:

```
def basic_die(
              size = (10000, 10000),
              street_width = 100,
              street_length = 1000,
              die_name = 'chip99',
              text_size = 100,
              text_location = 'SW',
              layer = 0,
              draw_bbox = True,
              bbox_layer = 99,
              ):
    D = Device(name = 'die')
    sx, sy = size[0]/2, size[1]/2
    xpts = np.array([sx, sx, sx-street_width, sx-street_width,
        sx-street_length, sx-street_length])
    ypts = np.array([sy, sy-street_length, sy-street_length,
        sy-street_width, sy-street_width, sy])
    D.add_polygon([xpts,ypts], layer = layer)
    D.add_polygon([-xpts,ypts], layer = layer)
    D.add_polygon([xpts,-ypts], layer = layer)
    D.add_polygon([-xpts,-ypts], layer = layer)

    if draw_bbox is True:
        D.add_polygon([[sx,sy], [sx,-sy],[-sx,-sy],[-sx,sy]], layer = bbox_layer)
    D.center = (0,0)
    t = D.add_ref(text(text = die_name, size = text_size, layer=layer))

    d = street_width + 20
    if type(text_location) is str:
        text_location = text_location.upper()
        if text_location == 'NW':
            t.xmin, t.ymax = [-sx + d, sy - d]
        elif text_location == 'N':
            t.x, t.ymax = [0, sy - d]
        elif text_location == 'NE':
            t.xmax, t.ymax = [sx - d, sy - d]
        if text_location == 'SW':
            t.xmin, t.ymin = [-sx + d, -sy + d]
        elif text_location == 'S':
            t.x, t.ymin = [0, -sy + d]
        elif text_location == 'SE':
            t.xmax, t.ymin = [sx - d, -sy + d]
    else:
        t.x, t.y = text_location

    return D
```

jonnyfountain, 2020-03-13:
Thank you for the fast response @amccaugh. Works like a charm

closed 2020-03-14 by amccaugh

amccaugh, 2020-05-06:
@jonnyfountain This has been fixed in version 1.3.0, thanks again for the bug report 

Also be sure to check out the new feature, `pg.packer()`
https://phidl.readthedocs.io/en/latest/#packer-align-distribute

`pg.packer(D_list, spacing = 1.25, aspect_ratio = (2,1))`

[image]