amccaugh/phidl #68: np.ceil return value not safely castable to int

gyger, 2020-05-04:
In routing.py the return value can not be safely cast to an int, leading to a bug in an up-to-date python stack (I am not sure when it became mandatory).
An easy fix is to explicitly cast the points for linspace in _arc. e.g.
`int(np.ceil(abs(theta)/angle_resolution))` 

The error seen in the console will be:  `object of type <class 'numpy.float64'> cannot be safely interpreted as an integer`

amccaugh, 2020-05-04:
Thanks for finding this bug @gyger .  I'm working on one last feature for the latest version, I'll fix this and include it in the update which should be out in a day or two.

amccaugh, 2020-05-06:
This has been fixed in version 1.3.0, thanks again for the bug report @gyger 

Also be sure to check out the new feature, `pg.packer()`
https://phidl.readthedocs.io/en/latest/#packer-align-distribute

`pg.packer(D_list, spacing = 1.25, aspect_ratio = (2,1))`

[image]

closed 2020-05-06 by gyger