amccaugh/phidl #70: Placing functions - align, distribute

gyger, 2020-06-26:
Hej,
thanks for this new functions, also the packer (although I have not used that one yet, I somehow too much OCD to allow the computer to scatter my devices).

I was thinking the align function in Inkscape has this possibility to treat the selection as group, I think that would be also super nice if this could be an parameter. So align(center) does not put them on top of each other, but moves the whole group of devices in the center of the parent.

The second thing I was wondering about is distribute, if we can extend it to also allow distributing on a grid, e.g. by choosing xy as a parameter. I am unsure how to efficiently parametrize that.

amccaugh, 2020-06-26:
Hi @gyger , glad to hear you've had the chance to explore the new functions.  

- For the align function, I'm not entirely sure what you're describing.  Can you post a picture of an approximate example of what you mean (maybe just using shapes in Inkscape)?
- I agree about the being able to place things on a grid, we're actually working on a couple functions to do exactly that (but first I have to finish the waveguide module so it might be a few weeks).  For the moment, you can use a for loop but it's pretty annoying to have to write everytime so a convenience function would definitely be good 👍 .  We also have a function that we're planning to add next month called autoarray() that lets you specify a Device-creating function, a default parameter set, and parameters to sweep, like so:
```
default_args = dict(theta = 45, start_angle = 0, angle_resolution = 2.5, layer = 7)

D =  autoarray_xy(
        function = pg.arc,
        default_args= default_args,
        arg_variations_x = {'width' : [0.2,0.25,0.30,0.35,0.5]},
        arg_variations_y = {'radius' : [1,2,4,6]},
        rotation = 0,
        dx = 100,
        dy = 100,
        align = 'ymax',)
```

gyger, 2020-06-27:
Thanks also for the outlook. I made a [rectangular placing function yesterday](https://gist.github.com/gyger/cad3292e7b920619c64dc549f0eb31b7) then, perhaps it can be useful for you or somebody else.

About the align, find a sketch below. If it does not make the thing clear, I will try to extend it more or I can just send you another gist.
[image]

This waveguide module sounds also very interesting (is it the open pull request or offline developed), I am currently using PICwriter for this, which fortunately also uses gdspy so I can combine it. If you need help with the wg module let me know.

amccaugh, 2020-06-27:
I like that grid() function quite a bit, would you want to make a pull request to add it to geometry.py?  (I'm happy to add it myself of course, but it's always nice to make sure the original author gets the credit on github for contributions)

I see what you mean about aligning multiple items.  This is actually in line with something I've been thinking about recently -- it seems to me what would be the most useful/intuitive/general is to be able to easily group items into a "selection" similar to how you would in Inkscape using the mouse.  Then once you have that group/selection, you can manipulate it in the usual ways (xmin = 5, center = (0,0) etc).  How does something like this strike you from a usability/intuitiveness perspective?

```
D = Device()
e1 = D.add_ref(pg.ellipse())
e2 = D.add_ref(pg.ellipse())
e3 = D.add_ref(pg.ellipse())

# Exlicitly create the group then manipulate it
my_phidl_group = e1+e3
my_phidl_group.center = (4,5)
my_phidl_group.rotate(25)

# Or for shorthand
(e1+e3).center = (4,5)
```

amccaugh, 2020-07-23:
@gyger What do you think about changing the input variables to be more consistent with how distribute() already works? 

Specifically:
```
grid(device_list,
     spacing = (50,100),
     shape = (6,4),
     separation = True)
```

Here, if `separation==False`, then the devices will be put on a fixed grid of dx=50, dy=100 (regardless of how large or small the devices are).  However, if `separation == True`, then the function enforces a separation between the columns of 50 units

[image]

gyger, 2020-07-24:
Sounds like a good thing, especially if it is consistent with distribute. :+1:

amccaugh, 2020-08-20:
These have all been added and will be in the next release!

closed 2020-08-20 by amccaugh