amccaugh/phidl #73: First feedback on the waveguide work

gyger, 2020-08-20:
I have seen with excitement that the wave-guide module is coming along. I wanted to add some initial feedback below, of course everything is up to you and I am aware that there is probably still a lot of movement happening at the moment.

- Paths should be connectable with +, I think. So I am wondering if we should add the group operator from + to | (for bitwise or), which also somehow reflects that both given elements are considered in the group, or we can overload the + in the Path operator as connecting path which is not so nice.
- XSection, the name is not intuitive.
- Could the XSection be part of the Port, always applied inward to the device?

Some more things to think of:
- Should one track the path as a subset of paths with circular information (OASIS supports circles, and some e-beams have circle elements), so perhaps the flattening into points should only happen at the end?

I will play a little bit more with the new parts and extend this "bug" report.

amccaugh, 2020-08-20:
Thanks for taking a look at the new features @gyger.

1) For the case of Path, I definitely agree it'd be much better not to have to have to type out `append` a hundred times when you're making a waveguide.  That said, I'd like to keep "+" as the operator for making Groups (I want to encourage people to use it, and `|` is unfortunately not as familiar to novice programmers).    I was thinking it might not be a bad idea to use the `+=` operator for Paths?  There's also a minor optimization that occurs when you use in-place operators like `+=` versus not-in-place operators like `+`, though it probably wouldn't make too much a difference here.  Would something like the following work?

```
P = Path()
P += pp.arc(radius = 10, angle = 90)
P += pp.arc(radius = 10, angle = -45)
P += pp.straight(length = 15)
```

2) I agree about `Xsection` not being a great name but I haven't been able to come up with any good alternatives!  If you have any suggestions I'd be happy to hear them.  I've looked at `Crosssection` and `CrossSection` but those don't look any better imo.  I like `Profile` but it might not be as semantically clear? At the very least, `xsection` is used in [klayout](https://github.com/klayoutmatthias/xsection) and [nazca](https://nazca-design.org/implement-a-cross-section/).  It's not much, but consistency is useful.

3) What were you thinking of in terms of advantages to have Xsection be a part of Port?  For me it makes the concept a little more confusing, but I might be missing what you're thinking of.

4) Keeping track of circular information would be neat, but would require substantial changes to basically the entire codebase!  Also, I suspect users will mostly be using Euler curves for their waveguide bends which are non-circular, making the optimization pretty minimal

atait, 2020-08-20:
Hi @gyger 

There might be some confusion between `XSection` and `WG_XS`. They are keeping the same information. The latter has methods that effectively create paths commonly used in photonics (e.g. an Euler bend) and then make that Device. "Cross-section" is just an industry standard, like @amccaugh says. I'm not sure what you mean by "smooth."

Tying a cross section to a port is a good idea! You can do it on your own like
```python
path = Path(...)
xs = Xsection(..., port_names=(1,2))
D = path.make(xs)
D.ports[1].info['XS'] = xs
```
The `.info` attribute cannot make it through GDS conversion – it has to exist in the same python process. For those reasons, I think the user should be responsible for how they track their Xsections instead of the library doing it.

Curve representations are very desirable in photonics layout tools. There are file formats that have curve elements, but they are relatively new, often proprietary, and understood by only very modern and specific lithography tools. The layout software is built around the file format. Without a curve file format, there is not a whole lot to be gained by keeping a curve representation in phidl because they all have to end up as polygons in the end.

I am glad you bring up OASIS. It is much, much better than GDS in every way. Unfortunately, I don't know of a python program for reading/writing OASIS like there is `gdspy`. I did make a hacky version [here](https://github.com/atait/phidl/tree/oasis) using klayout-in-phidl if you're interested. Use that if you're concerned about lithography software performance (e.g. Beamer, Calibre), file size on disk (e.g. for emailing), and/or memory usage when inspecting the layout in a viewer.

If you are concerned about speed and memory usage while building the layout, I recommend looking into the `device_lru_cache` decorator. When routing waveguides, you're going to have hundreds of identical 90º waveguide bends, so you effectively just make one `Device` for that first time and reuse it. Secondly, do your development in low-resolution like this
```python
import phidl.waveguides
phidl.wavegudes.minimum_bent_edge_length = 0.5
```
see here
https://github.com/amccaugh/phidl/blob/ebe8e688a1e3b56a3a9051efaa2d09b56dc7895a/phidl/waveguides.py#L20
Then up the resolution for the final version.

gyger, 2020-08-20:
Thank you for this long replies.

>     1. I was thinking it might not be a bad idea to use the `+=` operator for Paths?

I would expect += and + to behave the same. So I think that would be not such a good idea.

2) Profile was also something I was thinking of or something connected to Extrusion or Slice or Face. But perhaps XSection is the best available one. 
3) A cross-section can only change on a port and is linked to a port, so I thought the port could know what cross-section it is representing and therefor a port and the cross-section could be linked, but perhaps this is not general enough to allow for all the use-cases.
4) Yeah I guess circular elements are still to far in the future for the industry that moves quite slow. 👍 

> `XSection` and `WG_XS` are keeping the same information.

Yes, is the plan to make WG_XS superfluous using the Path and XSection? I assume so or?

About the speed, I am currently not yet concerned, although the file size would be much much smaller when storing bends instead of pixels (euler bends of course are an exception). Caching things clearly is a requirement. I also think once the gds is written, most of the information is lost. So I use only one way: Code -> GDS, not the other way around.

After thinking a little bit more, I would suggest moving the extrusion away from paths and into the Cross-section.
The path in principle could even be stored in the GDS, although perhaps that is not a useful thing to keep. The profile for the extrusion could also be saved in the gds as a cell that is "1D", then extruded along the path.

```
path = Path(...)
xs = Xsection(...)
D = xs.extrude(path, port_names=(1,2))
```

Perhaps I am missing something. Thanks for all your work on this library.

amccaugh, 2020-08-20:
1) I think you're right about the `+` operator, it's confusing if it's not consistent.  What do you think about the `&` operator for creating Groups @atait @gyger ?  For instance `mygroup = t1 & t2 & t3 & t4` ?   I agree the `|` operator makes sense from a boolean-logic perspective, but I have the impression that the `&` operator makes more sense semantically (for instance, you'd never say "I'm moving the blue square OR the red square OR the green square" if you were talking about moving all 3 squares at once)
2) I like "Slice" ... but maybe that's too confusing considering Python's built-in `slice()`.  "Face" generally refers to 2D shapes so I think that's out.  Naming these things is tough :/

For other notes
- Yes you're correct, WG_XS was the initial implementation but is no longer used
- The speed of this is very, very fast thanks to exclusively using C-optimized numpy functions -- tens of thousands of points can be computed in a millisecond or two
- My thought for doing `P.make(xsection)` was that the most common use case would be someone who had different waveguide cross-sections wanting to make multiple devices for different wavelengths using the same path.  i.e. `P.make(xsection_1550nm)` and then later `P.make(xsection_1064nm)`.  That said, there's nothing special about doing one or the other!  We could also add an operator so that we make a device from path `P` and xsection `X` like this: 

`D = P*X`

Thoughts?

amccaugh, 2020-09-03:
@gyger Would you mind taking a look at https://phidl.readthedocs.io/en/dev/tutorials/waveguides.html#Building-Paths-quickly ?  I think I've come up with an elegant solution to assembling Paths quickly and easily (and keeping `+` operator dedicated to making Groups)

gyger, 2020-09-05:
Nice. I worked on my first chip with waveguides using PHIDL dev-head. It is nice. I looked at building paths quickly mentioned above.

1. Appending a list works nicely and love the support of that.
2. The constructor of a Path could already accept that list, that would make two lines only one.
3. Using the extrude I found it very useful and worked immediately and i think the workflow works very well.

I think it could be nice if the CrossSection is like a LayerSet a dictionary or could be named. At least in my use case I tried to change the width of one element later, and now used the list index.

gyger, 2020-09-05:
One question on Positive resists, would you use just two elements in the cross section and offset them or use boolean magic later?

amccaugh, 2020-09-08:
- Yes I agree the constructor could save a line, that's a good idea.  Like `P = Path( pp.arc(radius = 10, angle = 90) )`
- Sounds good on the naming of the CrossSection.  Something like `X.add(width = 1, offset = 0, layer = 0, name = 'etch')`

With respect to positive resists, I'd generally use the boolean function `pg.outline()` afterwards.  It's possible to use multiple elements in the cross-section but often the ends are still open afterwards which is difficult/annoying to deal with.

amccaugh, 2020-09-08:
All functionality has been added.  We're almost ready for release!

closed 2020-09-08 by amccaugh