amccaugh/phidl #78: casting issue when moving components with labels

joamatab, 2020-10-05:
Hi Adam, 

I noticed this casting bug

```
import phidl.geometry as pg
from phidl import quickplot as qp
from phidl import Device

c = Device()
c << pg.ellipse(radii = (10,5))
c.add_label('hi', position=(0, 0))
c.movex(10.)

```
```
UFuncTypeError: Cannot cast ufunc 'add' output from dtype('float64') to dtype('int64') with casting rule 'same_kind'
```

the casting issue can only be solved by using position and movements with the same data types. For example, it works if position and movement are both `int`

``
c.add_label('hi', position=(0, 0))
c.movex(10)
``

or `float`

```
c.add_label('hi', position=(0., 0.))
c.movex(10.)
```

amccaugh, 2020-10-05:
Wow, nice find Joaquim -- I can't believe I haven't run into this before.  Got it fixed, thanks for the report!

closed 2020-10-05 by amccaugh

joamatab, 2020-10-06:
thank you for the fix Adam and for your latest phidl improvements ^^