amccaugh/phidl #91: add plt.ioff() option to quickplot2()

giumc, 2021-04-07:
Hi!
I would like to block script when quickplot() is called, until figure is closed.

If I run a .py script with the following code:

```import phidl.geometry as pg
from phidl import quickplot as qp
import matplotlib.pyplot as plt
r=pg.rectangle()
qp(r)
```

the figure immediatly disappears.
It seems that i can pause for `x` time the script via `plt.pause(x)' but that doesn't really do the job for me.

I would like qp() to block script until window is closed, as in 
```
import matplotlib.pyplot as plt
plt.ioff()
plt.figure()
plt.show()
```

Adding `plt.ioff()` before `qp()` doesn't work for me.
Is this something you could implement? is this already implemented?

amccaugh, 2021-04-07:
This is interesting, thank you for the concrete examples. I'll have to take a look through the code but I don't see why we shouldn't be able to make it blocking.  I've also been considering making a `set_quickplot_options()` function which allows you to set the options for the plot (such as showing/hiding subports, etc).  This would fit nicely in there, perhaps as `set_quickplot_options(blocking = True)`.

Give me a few days to take a look at it, I believe we can come up with something

amccaugh, 2021-04-08:
@giumc I think I've got a working implementation with blocking.  Can you pull the latest changes on the `dev` branch and install them and try the following code?

```
from phidl import quickplot as qp, set_quickplot_options
import phidl.geometry as pg

set_quickplot_options(blocking=True)
print('start')
qp(pg.compass())
print('middle')
qp(pg.snspd())
print('end')
```

giumc, 2021-04-09:
fantastic! works nicely. Kudos!

closed 2021-04-09 by giumc