amccaugh/phidl #92: show_subports=True in quickplot

giumc, 2021-04-08:
Hi again
I noticed that when creating higher lever hierachy of cells , ports are not printed in quickplot()
Here is a test function: 
```
import phidl.geometry as pg
from phidl.device_layout import Device,Port
from phidl import quickplot as qp
import matplotlib.pyplot as plt

r=pg.ring()

d=Device()

d<<r

r.add_port(Port('center',midpoint=(0,0),width=10))

r2=d<<r

r2.move(origin=(0,0),destination=(20,0))

d_super=Device()

d_super<<d

qp(d)

plt.pause(2)

qp(d_super)

plt.pause(2)

print('Ports in subcell',* d.get_ports(),sep='\n')
print('Ports in supercell',* d_super.get_ports(),sep='\n')
```
I tried replacing quickplot with quickplot2, but python3 crashes.
I read in the docs that there is a kwarg show_subports=True but it seems not to show subports.

amccaugh, 2021-04-08:
Note that the green ports displayed in quickplot *are* the subports.  Device-level ports are shown in red, ports contained in the device's references are "subports".  So in the above code, the Device `r` has a port called "center".  When `r` is referenced inside `d`, then `d` ends up containing two "subports" -- these are the ports defined in `r` but only *referenced* in `d`. In my modified code below all I've done is additionally quickplot `r`, and you see that there's a red port -- that's because the port is local to the Device `r`. 

```
import phidl.geometry as pg
from phidl.device_layout import Device,Port
from phidl import quickplot as qp
import matplotlib.pyplot as plt

r=pg.ring()

d=Device()

d<<r

r.add_port(Port('center',midpoint=(0,0),width=10))

# ADDED LINE HERE
qp(r)

r2=d<<r

r2.move(origin=(0,0),destination=(20,0))

d_super=Device()

d_super<<d

qp(d)

plt.pause(2)

qp(d_super)

plt.pause(2)

print('Ports in subcell',* d.get_ports(),sep='\n')
print('Ports in supercell',* d_super.get_ports(),sep='\n')
```

[image]

I should note that "subports" are only shown one-level deep.  That's because when you're interacting with a device, say `d` in the above example, you often want to position the references according to their ports.  But very rarely does it make sense to manipulate a sub-sub-port.  For instance, in `d_super`, if you wanted to interact with the sub-sub-port you'd have to access the reference `d`, and then inside `d` access the reference to `r`.  In that case, it makes more sense to just access the `d` Device object directly and manipulate the reference to `r` inside it.

giumc, 2021-04-08:
That makes sense, thanks for the clarification! 

I was misinterpreting the representation of Ports, now in hindsight this is the natural way of referencing and passing them across hierarchy.

Big Kudos for this layout tool, I think it really stands out wrt gdspy basic tool...

Thanks for the detailed explanation, hope it can be of help for others as well...

closed 2021-04-08 by giumc