amccaugh/phidl #93: pp.smooth() with collinear segments breaks

giumc, 2021-04-12:
When building paths from pp.smooth(), it seems that the code breaks when collinear points are passed.

As a MWE, this code works nicely:
```
import phidl.path as pp

from phidl import quickplot as qp

path1=pp.smooth([(0,0),(50,0)])
import phidl.path as pp

from phidl import quickplot as qp

path1=pp.smooth([(0,0),(50,0)])

qp(path1)

```

However, if another point on the x axis is passed, the code breaks :

```
import phidl.path as pp

from phidl import quickplot as qp

path1=pp.smooth([(0,0),(50,0),(100,0)])

qp(path1)

```
 
I am trying to implement an autorouting script that would greatly benefit from this , let me know if there is any chance you can implement this!

amccaugh, 2021-04-12:
Great bug find, can't believe I didn't run into that yet!  We can definitely correct for this by pre-filtering the points list to remove these redundant points.  Let me take a look and hopefully I'll have a fix in a couple days

amccaugh, 2021-04-13:
@giumc Can you try the latest commit on the `dev` branch?  I believe this should be fixed now.

Also, if you have the opportunity I've been making some changes to `quickplot()` -- the mousewheel should now work in interactive windows.  If you have the chance to try it out I'd appreciate it.  If you're using a notebook-like system, note you'll have to put something like `%matplotlib qt` in the beginning.

giumc, 2021-04-14:
@amccaugh installed `phidl` in branch `dev` at commit `a612940` . 
I get error :
```
line 139, in set_quickplot_options
    _quickplot_options['blocking'] = blockin
NameError: name 'blockin' is not defined
```

the error is solved if I commit back to where the set_quick_options() was introduced (i.e.`f855b19')

let me know if I can test out some other thing ( and i apologize if this is not the right way of reporting bugs )

amccaugh, 2021-04-14:
This is exactly how to do it -- I appreciate the responses.

Looks like that was just a silly typo, I pushed a commit that should fix it

giumc, 2021-04-14:
Bug fixed! The mouse works smoothly for me, I will let you know if something related to mousewheel comes up...
Thanks !

closed 2021-04-14 by giumc