amccaugh/phidl #95: Extruded Paths geometric hash error

joamatab, 2021-04-17:
Hi Adam,

how can we get the geometric hash from a Device that comes from an extruded path?

I get an error when I do this

```
from phidl import CrossSection
import phidl.path as pp

X1 = CrossSection()
X1.add(width=1.2, offset=0, layer=2, ports=("in1", "out1"))
X1.add(width=2.2, offset=0, layer=3)
X1.add(width=1.1, offset=3, layer=1)

P3 = pp.straight(length=15)
c = P3.extrude(X1)
print(c.hash_geometry())
```

amccaugh, 2021-04-18:
Thanks for the bug report - this was a strange one.  I pushed a bugfix to `dev`, can you check to see that it works on your end?

amccaugh, 2021-04-27:
This should be addressed in #96 , let me know if you run into any further problems

closed 2021-04-27 by amccaugh

joamatab, 2021-04-30:
yes Adam, it works great on the new phidl 1.5.0

thank you for your super quick fix ^^