amccaugh/phidl #98: smooth feature request

joamatab, 2021-04-30:
Hi Adam,

I've reallly like the simplicity of defining routes with smooth as well as the new CrossSection.
I am thinking about simplifying the routes in gdsfactory with the new smooth function

Something that is important for photonic routes is that you want to keep the idea of a component (bend, straight) because for circuit simulations a bend and a straight will have different models

What do you think of breaking smooth into a dict of straights and bends?
How would you detect the straight paths in `pp.smooth`?

```
import numpy as np
from phidl import Device
import phidl.geometry as pg
import phidl.path as pp

def smooth(
    points,
    radius=4.0,
    bend_path_function=pp.euler,
    **kwargs,
):
    """Returns a list of Paths from a series of waypoints. Corners will be rounded
    using `bend_path_function` and any additional key word arguments (for example,
    `use_eff = True` when using `bend_path_function = pp.euler`)

    FIXME: need to extract straights

    Args:
        points: array-like[N][2] List of waypoints for the path to follow
        radius: radius of curvature, passed to `bend_path_function`
        bend_path_function: function that controls how the corners are rounded.
        **kwargs: Extra keyword arguments that will be passed to `bend_path_function`
    """
    straights = []

    points = np.asfarray(points)
    normals = np.diff(points, axis=0)
    normals = (normals.T / np.linalg.norm(normals, axis=1)).T

    dx = np.diff(points[:, 0])
    dy = np.diff(points[:, 1])
    ds = np.sqrt(dx ** 2 + dy ** 2)
    theta = np.degrees(np.arctan2(dy, dx))
    delta_theta = np.diff(theta)
    delta_theta = delta_theta - 360 * np.floor((delta_theta + 180) / 360)

    # Create arcs
    bends = []
    radii = []
    for dt in delta_theta:
        P = bend_path_function(radius=radius, angle=dt, **kwargs)
        chord = np.linalg.norm(P.points[-1, :] - P.points[0, :])
        r = (chord / 2) / np.sin(np.radians(dt / 2))
        r = np.abs(r)
        radii.append(r)
        bends.append(P)

    d = np.abs(np.array(radii) / np.tan(np.radians(180 - delta_theta) / 2))
    encroachment = np.concatenate([[0], d]) + np.concatenate([d, [0]])
    if np.any(encroachment > ds):
        raise ValueError(
            "[PHIDL] smooth(): Not enough distance between points to to fit curves."
            " Try reducing the radius or spacing the points out farther"
        )
    p1 = points[1:-1, :] - normals[:-1, :] * d[:, np.newaxis]

    # Move bends into position
    new_points = list()
    new_points.append([points[0, :]])
    for n, dt in enumerate(delta_theta):
        P = bends[n]
        P.rotate(theta[n] - 0)
        P.move(p1[n])
        new_points.append(P.points)
    new_points.append([points[-1, :]])
    new_points = np.concatenate(new_points)

    # P = Path()
    # P.rotate(theta[0])
    # P.append(new_points)
    # P.move(points[0, :])
    return dict(straights=straights, bends=bends)

def get_route(paths_dict, cross_section):
    D = Device("route")
    bends = paths_dict["bends"]
    straights = paths_dict["straights"]

    for i, path in enumerate(bends):
        bend = path.extrude(cross_section)
        bend.name = f"bend_{i}"
        D.add_ref(bend)

    for i, path in enumerate(straights):
        straight = path.extrude(cross_section)
        straight.name = f"straight_{i}"
        D.add_ref(path.extrude(cross_section))
    return D

if __name__ == "__main__":
    from phidl import Path, CrossSection
    from phidl.quickplotter import quickplot as qp

    dx = 20
    dy = 20

    paths_dict = smooth([(0, 0), (dx, 0), (dx, dy), (dx + dx, dy)])
    X = CrossSection()
    X.add(width=1, offset=0, layer=0, ports=("in", "out"))

    route = get_route(paths_dict, cross_section=X)
    qp(route)

```

amccaugh, 2021-05-04:
Hi @joamatab , in its current implementation I'm not really sure it blends well with the underlying principles of how `phidl` is structured.  For instance, *everything* in `phidl.path` returns a Path, and *everything* in `phidl.geometry` returns a Device.  This is intended to keep things intuitive -- the change you're proposing breaks this and the [PLOA](https://en.wikipedia.org/wiki/Principle_of_least_astonishment).

Could you give me a little more info about your needs?  For instance, would it suffice if there was some information embedded inside the output of `smooth` that allowed you to extract the individual Paths?  What other information do you need access to that's not currently provided?

joamatab, 2021-05-04:
Hi Adam,

For automatic routing of port bundles it's nice to be able to get the delay
of each route, and the index of bends is different from straights
How about creating a new function that returns a paths into sub paths?

Allowing Bézier bends where there is not enough space, as well as
supporting 180 degrees Euler bends would be great

Let me know what you think
Joaquín

amccaugh, 2021-05-26:
I think adding a Bezier bend option is a good one, and I think we can create an easy-to-use `CompoundPath` or something like it that allows you to separate out bends from straights

amccaugh, 2021-05-26:
Also, to clarify -- are you interested in making the bends a different cross-section than the straights?  Because if not, you could use the [curvature calculation](https://phidl.readthedocs.io/en/dev/tutorials/waveguides.html#Curvature-calculation) to very easily extract the curvature of every point along the path, then use that curvature to compute the effective index

joamatab, 2021-05-26:
Im interested in having meaningful properties for bends and straights, so they can have different names, properties or circuit models associated with them.

let me know what you think

amccaugh, 2021-05-26:
That makes sense, just wanted to see if your problem could be resolved with existing code

closed 2021-06-23 by amccaugh