amccaugh/phidl #99: Rename cells?

giumc, 2021-05-08:
Straight question, can you rename cells? 
If so, would you propose a workaround to accomplish this?
I see that there is an "_internal_name" protected variable, I would like not to touch it ( I would assume there is a reason why it's protected).
Let me know

Giuseppe

amccaugh, 2021-05-11:
I think that should definitely be doable. Let me take a look and I'll try to get back to you in the next couple days. 

I suspect there's actually some deprecated code in there that I can remove -- I believe _internal_name is not actually used (with gdspy it used to be easier to guarantee all cells had unique names in advance, but I think that's since changed)

amccaugh, 2021-05-13:
@giumc could you try loading the latest commits on `dev` and try it out?  I have removed `D._internal_name`, which was bookkeeping needed in earlier versions of gdspy (but is no longer necessary). You should be able to modify names just be doing

`D.name = 'newname'`

giumc, 2021-05-20:
apologies, I forgot to reply! 
It works like a charm, thanks!

closed 2021-05-20 by giumc