amccaugh/phidl #100: boolean operations on layers?

petebarry1988, 2021-05-15:
Is it possible to perform boolean operations on Layers, rather than Devices (i.e. similar to the functionality in KLayout) ? I tried to create and pass phidl.Layer objects when creating Devices, but that didn't seem to have the desired effect. Apologies if this is already possible and I've just missed it (I couldn't find it in gdspy either) .

```
import phidl, phidl.geometry as pg
from phidl import quickplot as qp

l0 = phidl.Layer( 0, name = "layer0" ) 
l1 = phidl.Layer(  1, name = "layer1" ) 

r0 = pg.rectangle( layer = l0 )
r1 = pg.rectangle( layer = l1 ).movex(1)

qp( pg.boolean(A = r0, A = r1, operation = 'xor', layer = 2) )  # expected result
qp( pg.boolean(A = l0, B = l1, operation = 'xor', layer = 2)  ) # nothing there

```

amccaugh, 2021-05-15:
Not in precisely the same way as KLayout -- but there are functions which will accomplish what you're looking for.  There's a function called `pg.extract(device, layers)` which will allow you to accomplish this.  What I would do in your case is add everything to a temporary Device, then run `pg.extract` (which will create a new device with only the layers you want on them).

```
# continuing from your code above

# Add relevant geometry to a temporary device D
D = Device()
D << r0
D << r1

# Create new devices with only the desired layers
D_layer0 = pg.extract(D, layers = [0])
D_layer1 = pg.extract(D, layers = [1])

# Perform boolean
pg.boolean(A = D_layer0, A = D_layer1, operation = 'xor', layer = 2)
```

We need to flesh out the documentation for Layers, I'll try to include an example of this when we do.  For now there's a crude example here:  https://phidl.readthedocs.io/en/latest/geometry_reference.html?highlight=layers#Copying-and-extracting-geometry.

Out of curiousity, what is your use case here?  I've seen that functionality in KLayout but never had an occasion to use it myself.  (The reason I ask is typically most of what I see people do is based around operations within a single Cell/Device, not across multiple ones)

amccaugh, 2021-05-19:
Also just FYI, I wrote up a tutorial for using Layers effectively:

https://phidl.readthedocs.io/en/dev/tutorials/layers.html

petebarry1988, 2021-05-20:
That's great - thanks for the quick reply. I'll try this out, but its look like this should work (so i'll close the issue).

In this instance, my specific use case is somewhat trivial and is just being used to label a unique ID for each die on a wafer. I create a dedicated text layer for each device layer, and the pair of layers are then xor'd as the final operation. I'm sure that this could be achieved a number of different ways, but this seemed quite efficient (i.e. not having to create copies of Devices...etc.).

closed 2021-05-20 by petebarry1988