amccaugh/phidl #102: manhattan route withg gradual bend

joamatab, 2021-05-20:
Hi Adam,

I noticed that phidl.routing.route_manhattan has some issues if it does not have enough space for a route

```
from phidl import Device, Port
from phidl.quickplotter import quickplot as qp
from phidl.routing import route_manhattan

if __name__ == "__main__":
    c = Device()

    ports1 = [
        # Port("in1", (10, 5), 0.5, 90),
        # Port("in2", (-10, 20), 0.5, 0),
        # Port("in3", (10, 30), 0.5, 0),
        # Port("in4", (-10, -5), 0.5, 90),
        # Port("in6", (0, 0), 0.5, 0),
        Port("in5", (0, 0), 0.5, 0),
    ]

    ports2 = [
        # Port("in1", (90, -60), 0.5, 180),
        # Port("in2", (-100, 20), 0.5, 0),
        # Port("in3", (100, -25), 0.5, 0),
        # Port("in4", (-150, -65), 0.5, 270),
        # Port("in6", (0, 12), 0.5, 180),
        Port("in5", (15, 6), 0.5, 180),
    ]
    N = len(ports1)

    for i in range(N):
        route = route_manhattan(ports1[i], ports2[i], radius=3, bendType="gradual")
        c.add(route.references)

    # qp(c)
```

[image]

What do you think of adding a check and raise an Error if it does not fit?

```
    if bendType == "gradual":
        b = _gradual_bend(radius=radius)
        radius_eff = b.xsize
    else:
        radius_eff = radius

    if (
        abs(port1.midpoint[0] - port2.midpoint[0]) < 2 * radius_eff
        or abs(port1.midpoint[1] - port2.midpoint[1]) < 2 * radius_eff
    ):
        raise ValueError(
            f"bend does not fit (radius = {radius_eff}) you need radius <",
            min(
                [
                    abs(port1.midpoint[0] - port2.midpoint[0]) / 2,
                    abs(port1.midpoint[1] - port2.midpoint[1]) / 2,
                ]
            ),
        )
```

amccaugh, 2021-05-25:
Hi Joaquin, I agree an error would be helpful here -- would you mind making a pull request to `dev`?

joamatab, 2021-05-25:
great, thank you Adam!

see PR #107

PR #107 (2021-06-11): routing raise error if it does not fit
closes #102
closed 2021-06-03 by amccaugh