amccaugh/phidl #103: aliases behavior upon Device.add()

giumc, 2021-05-20:
Hi amccaugh! 
I come to you again looking for delucidations.

When using the ```Device.add()``` method, is the "destination" class supposed to store the ```alias``` for the "source" cell references?

I am trying to use such a functionality, in this example it seems that the ```d``` cell doesn't inherit the ```alias``` of the reference to ```r```...

```import phidl.geometry as pg
import phidl.device_layout as dl

r=pg.rectangle()

c=dl.Device(name='test')

c.add_ref(r,alias='r1')

d=dl.Device(name='testwrapper')

d.add(c)
print("Original Class\n")
print(c)
print("\nWrapper Cell\n")
print(d)
```

amccaugh, 2021-05-21:
Yes, this is intended behavior.  The reason is similar to how "subports" work -- for a given Device `D`, you can only access the aliases for objects in `D`.  The reason for this is to avoid name collisions.  Consider the following situation:

```
R = pg.rectangle()
Rcontainer = Device()
Rcontainer.add_ref(R, alias = 'myrect')

D = Device()
rc1 = D.add_ref(Rcontainer)
rc2 = D.add_ref(Rcontainer)
```

At this point, I now have two references to `Rcontainer` in my device `D`.  Both references contain the alias `'myrect'`.  Let's assume that `D` will inherit the aliases from the added references -- when I try to access`D['myrect']`, which one does it return, the alias from `rc1` or the alias from `rc2`?

amccaugh, 2021-05-21:
On a sidenote, we intended that no one would ever use the `.add()` function in PHIDL -- I've actually thinking we should remove access to it (or renaming it `_add()` so it's obvious it's not supposed to be used.  But let me ask you this before we do so -- do you have any use cases where you couldn't equivalently use `add_ref()`, `add_polygon()` or `add_array`?

giumc, 2021-05-21:
Thanks for the response, it makes a lot of sense.

I am using it to deep copy a Device, and add the copy to another Device. As I was looking for the right word to use to describe this, I noticed that there is a ```deepcopy``` method for ```Device()```, which probably does the same thing. i should probably use that instead!

amccaugh, 2021-05-21:
Make sure you use https://phidl.readthedocs.io/en/latest/API.html#deepcopy , that's the correct deepcopy

closed 2021-05-25 by amccaugh