amccaugh/phidl #112: Interactive ```qp``` with jupyter

giumc, 2021-06-19:
Hi ```amccaugh```.

I am trying to get interactive plots in ```jupyter lab``` cells. I am using your latest version of your ```dev``` branch ```phidl``` 
(Merge: 7a86876 c618648).
To get the interactive window directly on the ```jupyter``` cell , I am using the ```widget``` backend . 
If I use ```qt5```, I do get the interactive plots, but on a separate window which is annoying...
To install ```widget```, I use the tutorial [here](https://towardsdatascience.com/how-to-produce-interactive-matplotlib-plots-in-jupyter-environment-1e4329d71651).

When I run the following cell from a clean kernel
```
%matplotlib widget

from phidl import quickplot as qp
import phidl.geometry as pg

qp(pg.circle()
```
The interactive plot is correctly displayed.

If I then proceed to another cell where I write

```qp(pg.rectangle())```

I get no output.

My question is :
* Should I keep using ```widget``` or do you have another recommendation on the backend to pair with ```phidl``` ? 

* If I should use ```widget```, would you recommend a solution/strategy to overcome this issue?

amccaugh, 2021-06-23:
Do basic matplotlib commands work with interactive windows?  For instance, if you try the following, do you get an interactive window?  I haven't been able to get 

```
%matplotlib widget
from matplotlib import pyplot as plt
plt.figure()
plt.plot([1,2,4,7])
plt.figure()
plt.plot([1,2,4,7])
```

If that does in fact work, what I would suggest is setting [the quickplot option to create a new window each time](https://phidl.readthedocs.io/en/dev/API.html?highlight=set_quickplot_options#set-quickplot-options) to `True`:
```
from phidl import set_quickplot_options
set_quickplot_options(new_window = True)
```

giumc, 2021-06-23:
It does work for me! 
I think it needs the ```ipympl``` module to run smoothly.
Thanks!

closed 2021-06-23 by giumc

amccaugh, 2021-06-23:
Out of curiousity, do you have issues with the mousewheel/trackpad accidentally zooming in the interactive notebook?  I've noticed in some systems that trying to scroll in jupyter sometimes inadvertently activates the "scroll-to-zoom" feature of quickplot.  Have you experienced this?

giumc, 2021-06-23:
Yeah , that was going to be my next problem to fix... Whenever I scroll on an interactive window, both the interactive zoom and the jupyter page scrolling are activated. I don't know who to blame for this, whenever I get this fixed i will let you know!

amccaugh, 2021-06-29:
Could you try the latest commit on `dev`?  I believe I fixed the interactive window issue automatically.  (I also added an option to `set_quickplot_options` that lets you force `interactive_zoom = False`)