amccaugh/phidl #113: PHIDL won't update in anaconda navigator

georgiosauce, 2021-08-12:
Stuck on version 1.4.2 despite installing the latest version. Command 'conda list' shows version 1.5.2 but anaconda will only run 1.4.2 and will not show update available therefore cannot use gridsweep function

amccaugh, 2021-08-12:
Can you paste the commands you're using to install/check the version?  For instance

`pip install -U phidl` to install phidl

then inside python running

```
import phidl
print(phidl.__version__)
```

georgiosauce, 2021-08-13:
conda list shows version 1.5.2 in terminal window. import phidl -> print(phidl.__version__) in Jupyter notebook gives 1.4.2

georgiosauce, 2021-08-13:
Also I get this message despite phidl being listed in conda list

PackagesNotFoundError: The following packages are not available from current channels:

  - phidl

Current channels:

  - https://repo.anaconda.com/pkgs/main/osx-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

georgiosauce, 2021-08-13:
Got it to work, I was in the wrong directory when installing

closed 2021-08-13 by georgiosauce

amccaugh, 2021-08-13:
I see -- phidl is not uploaded to `conda`, so instead of using that you should do the following

1. Click Open terminal from environment
[image]
2. In the terminal run `pip install -U phidl`