amccaugh/phidl #122: Hierarchical Layout Question

giumc, 2021-12-09:
Hi @amccaugh !
I am trying to find a way to get relative position for ```DeviceReference```s nested in arbitrarily nested hierarchy.

I hope this scenario will clarify what I mean:
```
r=pg.rectangle()

p=dl.Device('parent')
rect_ref=p.add_ref(r,alias='rect_ref')

rect_ref.move(destination=(5,0))

gp=dl.Device('grandparent')
parent_ref=gp.add_ref(p,alias='parent_ref')

parent_ref.move(destination=(5,0))
```

Now, if I query for ```gp['parent_ref'].origin``` I will get ```[5 0]``` . However, if I type ```gp['parent_ref'].parent['rect_ref'].origin``` I will also get ```[5 0]```. While this makes perfect sense in a simple Parent```->```Reference tree, I wonder if you thought of a way of returning ```DeviceReference``` properties with respect to a more nested hierarchy. (To clarify, I am looking for ways to get ```[10 0]``` in this case)

I thought about it for a bit and I don't see any obvious way of achieving this with ```phidl``, so before coding it by myself I thought of checking in with you to see whether I am missing something here.

Thanks!

amccaugh, 2021-12-10:
Hi @giumc, we don't have that functionality for DeviceReferences but we do for Ports.  There's a (admittedly not-well-documented) function called `Device.get_ports()` which will do exactly what you're asking, except for ports.  If you're just interested in certain points-of-interest (for instance, the location of a specific feature in your device), you may be able to add a Port there and grab it later.  You can also filter it effectively using the `Port.info` dictionary.  See, for example:

```python
import phidl.geometry as pg
from phidl import Device
r=pg.rectangle()

p= Device('parent')
rect_ref=p.add_ref(r,alias='rect_ref')
port1 = p.add_port('point_of_interest', midpoint = (1.7,1.7), width = 0)
port2 = p.add_port('useless_port', midpoint = (0.5,0.5), width = 0)
port1.info['is_interesting'] = True

rect_ref.move(destination=(5,0))

gp= Device('grandparent')
parent_ref=gp.add_ref(p,alias='parent_ref')

parent_ref.move(destination=(5,0))

all_ports = gp.get_ports()

print(all_ports)
# Prints
# [Port (name point_of_interest, midpoint [6.7 1.7], width 0, orientation 45),
#  Port (name useless_port,      midpoint [5.5 0.5], width 0, orientation 45) ]

interesting_ports = [port for port in all_ports if 'is_interesting' in port.info]
print(interesting_ports)
# Prints
# [Port (name point_of_interest, midpoint [6.7 1.7], width 0, orientation 45)]
```

giumc, 2021-12-13:
That's awesome! I just realized not that was a bit of an [XY problem](https://xyproblem.info/) as what I wanted was actually getting the ports that come with certain ```DeviceReference``` that I need in top hierarchy. Thanks!

closed 2021-12-13 by giumc