amccaugh/phidl #127: Obstacles for Automated Routing

a1ward, 2022-03-02:
Hello,

I stumbled into this package while I was looking for a method that could be used to route from pads to device terminals automatically. First I wanted to thank you for all of this work, this is a really great idea and so far it's been a really great tool to learn how to use. I see so much potential with it!

So on to my "issue". I may have misread the "routing" documentation, but I had assumed that "obstacles" could be created and routed around using the built in phidl functions. After testing things out on my own, I've come to realize that this may not be the case, and that the "obstacles" in the documentation were simply meant to demonstrate how a user could manually tweak routing to avoid the obstacles placed in the automated routing's path.

I am essentially wondering if there is a function that I am missing that has the routing take obstacles into account when connecting port 1 to port 2? In the examples you have for simple xy routing, it works quite nicely, but generally speaking, the amount of manual intervention in the routing functions (i.e. creating a manual path per connection, or creating a large combination of "yyyxxyyy" like strings) makes using this just as time consuming as hand crafting polygons in a layout editor when the layout increases in complexity. If there is no way of automatically doing this as of now, are there any future plans to incorporate automatic recognition of "obstacles" in the routing functions?

Thanks for your time/help!

amccaugh, 2022-03-08:
Hi there, glad to hear you're finding phidl useful.  You're absolutely correct -- the routing done around obstacles is basically a manual process.  There's actually quite a lot of work done in this area, and the generic term they're called are "autorouters".  It wouldn't be impossible to implement, but it may be a little beyond the scope of phidl.  For instance, take a look at the list of available algorithms [here](https://en.wikipedia.org/wiki/Routing_(electronic_design_automation)).  Additionally, they take quite a lot of finessing to use properly (see for instnace [all the tricks qrouter uses](http://opencircuitdesign.com/qrouter/), and even then people generally have lots of [complaints about the results](https://forum.kicad.info/t/why-do-people-choose-to-not-use-the-auto-router/25849/14).  Overall, I like the idea but given my lack of expertise in this area, I'd be very hesitant to do it unless there was a very fast/well-documented library that handled a lot of these concerns.  Hopefully that answers your question

a1ward, 2022-03-10:
Thanks so much for this additional information! I'll spend some time looking through these different algorithms for routing and perhaps post again on this page if anything comes up that might be worth implementing.

I really just wanted to make sure that I wasn't missing something in the built in routing functions of phidl. Thank you so much for your time, and all the hard work here.