amccaugh/phidl #136: device.remove() doesn't traverse its references

basnijholt, 2022-03-11:
The following:
```python
device = phidl.Device()
box = phidl.geometry.bbox()
ref = device.add_ref(box)
device.remove(ref.parent.polygons[0])
```
raises
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
File ~/mambaforge/envs/qms-lab-20220224-main/lib/python3.9/site-packages/phidl/device_layout.py:1556, in Device.remove(self, items)
   1555 if isinstance(item, gdspy.PolygonSet):
-> 1556     self.polygons.remove(item)
   1557 if isinstance(item, gdspy.CellReference):

ValueError: list.remove(x): x not in list

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
Input In [35], in <module>
      2 box = phidl.geometry.bbox()
      3 ref = device.add_ref(box)
----> 4 device.remove(ref.parent.polygons[0])

File ~/mambaforge/envs/qms-lab-20220224-main/lib/python3.9/site-packages/phidl/device_layout.py:1563, in Device.remove(self, items)
   1561             self.aliases = { k:v for k, v in self.aliases.items() if v != item}
   1562         except:
-> 1563             raise ValueError("""[PHIDL] Device.remove() cannot find the
   1564                              item
   1565                              it was asked to remove in the Device:
   1566                              "%s".""" % (item))
   1568 self._bb_valid = False
   1569 return self

ValueError: [PHIDL] Device.remove() cannot find the
                                     item
                                     it was asked to remove in the Device:
                                     "Polygon (4 vertices, layer 0, datatype 0)".
```

I am not sure whether this is expected or whether I should expect remove to traverse down the references?

amccaugh, 2022-03-12:
This is expected -- `remove()` is not intended to search through the hierarchy, it's strictly meant to remove elements (labels, polygons, references, etc) from the Device itself.  The thinking is as follows:

Say I had two Devices, `D1` and `D2` which both had references to another device `D3` that contains polygon `poly`.  If `remove()` traversed the heirarchy, calling `D1.remove(poly)` would delete the polygon out of `D3`.  As a result, in addition to the polygon being missing from `D1`, it would now also be missing from `D2`.  At least in my opinion, this would violate the https://en.wikipedia.org/wiki/Principle_of_least_astonishment !

basnijholt, 2022-03-12:
Ok great, thanks for the explanation 😄

closed 2022-03-12 by basnijholt