amccaugh/phidl #140: support gdstk backend

tvt173, 2022-04-19:
Hi @amccaugh ,

I was curious if you have considered supporting gdstk as a new backend (substitute for gdspy). I think it is starting to become feature complete and on par with the original API and functionality of gdspy. Since I believe all new development is going into gdstk instead, it would be great to leverage all the great work that @heitzmann and others are putting in. Not sure if you had already considered this or looked further.

@joamatab

heitzmann, 2022-04-20:
As the main developer of gdstk, I'd just like to point out that there are still known bugs in gdstk that I'm working on (my time to do this has been quite limited in the last weeks, but I haven't vanished or anything). I'd also consider that the API won't be frozen before version 1.0, that is, breaking changes are possible between versions.
This is not to say I'm against it (quite the opposite, that would probably help identify more bugs), I'm just pointing out issues that could lead to frustration.

amccaugh, 2022-04-20:
Thanks for chiming in @heitzmann .  Right now my own time is pretty limited, so I don't have any immediate plans to migrate to `gdstk`(although I'm definitely interested in trying it out at some point).

@heitzmann On an unrelated note, have you thought about transitioning `gdstk` away from floating point to integer-math at any point?  That was one of the appeals of the `klayout` module to me, as floating point calculations are often troublesome (cause numerical problems that rounding can't fix, vary slightly between platforms, etc)

tvt173, 2022-04-20:
thanks @amccaugh 

understood... ok maybe once one of us gets some time, we can try it out 🙂

amccaugh, 2022-04-20:
Very interesting @heitzmann !  In terms of the rounding error, one thing I had briefly thought about was the fact that when you use a 64-bit integer to represent distances, it spans about 19 orders of magnitude (from ~1e-19 to 1e19).  If you assume an integer database unit of 1e-12 (e.g. 1 unit = 1 picometer), it would allow you to represent any object from 1 picometer to 10^7 meters, which I think one could safely state as covering the entire span of interesting length scales.  Afterwards, when saving to GDS, you'd round from 1e-12 to 1e-9 or whatever you wanted your final database size.  In that case, you might be able to get the best of all worlds:
- Rounding errors due to integer rounding are at most 1 picometer
- Extremely low likelihood of rounding errors when rounding to your final database units (eg nanometer)
- No issues with floating point implementations producing different answers on different systems
- Innate compatibility with clipper et al

heitzmann, 2022-04-20:
I believe you're right. Rounding errors would probably not be an issue if we fix the scale to 10⁻¹² or 10⁻¹⁵ even—the span would still cover 10 km of layout! There could be and issue with user code expecting to use floating point values… I'm sure a new user would be very confused by the magnitude of numbers they see. And the performance issue would exist, of course. But I'll keep the idea in the back burner… maybe it could be opted-in through a compilation flag? Thanks for the thoughts!

amccaugh, 2022-04-20:
Yes, it'd probably benefit from having a fixed-point implementation that shows the user decimals so you don't end up with ridiculous-looking numbers, but be internally stored as an integer.  Anyway, probably not going to worry about it for now but I'm glad you think the idea is interesting

closed 2022-05-15 by amccaugh