amccaugh/phidl #145: Ports are lost upon flattening

basnijholt, 2022-05-26:
See this example:
```python
import phidl

D = phidl.device_layout.Device()
D2 = phidl.device_layout.Device()
D2.add_port(phidl.Port("test"))
D << D2
D.flatten()

print(D.ports)
```
which prints an empty dict.

joamatab, 2022-05-26:
The way we handle this in gdsfactory is that we make a copy of the original `Component` and maintain the same ports

```
import gdsfactory as gf 

c1 = gf.components.straight()
c2 = c1.flatten()
print(c2.ports)
```

amccaugh, 2022-05-26:
Hmm, I think this might be a bug, but at the same time the example you're giving @joamatab is different than the one @basnijholt provided.

In your example, `c1` has some its own ports, and those are preserved after flattening. That *should* be the same as in phidl, but I believe it's not right now.  The case @basnijholt described is that ports from the **references within** `D` (i.e. `a` and `b` in my example code) are not copied into the newly-flattened Device

basnijholt, 2022-06-07:
@amccaugh, I see your point! There would not be a way to accomplish this with the current design. It might also not make sense to preserve ports upon flattening anyway.

Thanks for the explanation :)

closed 2022-06-07 by basnijholt