amccaugh/phidl #148: update conda package

joamatab, 2022-06-06:
A while ago i was able to publish the conda package for phidl, however it's out of date

How can we update it to the latest version?
https://github.com/conda-forge/phidl-feedstock

@willsALMANJ 
@proppy

https://github.com/gdsfactory/gdsfactory/issues/226

amccaugh, 2022-06-12:
Thanks for the report, I'm unlikely to have time to be able to learn the `conda` ecosystem, but if someone can suggest an easy way of submitting/updating the conda package I'd be game to keep it updated

wshanks, 2022-06-13:
@joamatab Did you change your username? The maintainer of https://github.com/conda-forge/phidl-feedstock is listed as @joaquin. If you used to control that account and don't any more, can you make a PR to the repo that changes the `recipe-maintainers` entry to @joamatab? Then we can ask one of the conda-forge maintainers to merge it to restore your merge rights for the feedstock (I usually ask in the gitter chat at https://gitter.im/conda-forge/conda-forge.github.io). Also, it sounds like you could add @amccaugh to the recipe maintainers list as well for helping to keep it updated.

joamatab, 2022-06-20:
perfect,
see https://github.com/conda-forge/phidl-feedstock/pull/5

how can we update to the latest version 1.6.1?
https://anaconda.org/conda-forge/phidl

joamatab, 2022-06-22:
Awesome!

the conda package is up to date now :tada:

closed 2022-06-22 by joamatab