amccaugh/phidl #166: pg.boolean returns polygons where properties are the same object in memory

basnijholt, 2022-11-04:
See:

```python
import phidl

a = phidl.geometry.bbox(bbox=[(0, 0), (1, 2)])
b = phidl.geometry.bbox(bbox=[(0, 0.5), (1, 1.5)])
d = phidl.geometry.boolean(a, b, operation="not")

phidl.quickplot([a, b])
set(id(p.properties) for p in d.polygons)
```
[image]
and prints:
```
{6076552704}
```
which means that 
```python
d.polygons[0].properties["a"] = 1
print(d.polygons[1].properties)
```
prints `{'a': 1}`.

@joamatab, maybe also happens in `gdsfactory`?

joamatab, 2022-11-07:
yes Bas, and at least in gdsfactory you cannot save the GDS after setting the properties of the polygons

https://github.com/gdsfactory/gdsfactory/issues/821

amccaugh, 2023-01-05:
I'm not really familiar with properties it seems -- what are they generally used for?  Some kind of dictionary I see, but are they present in the GDSII spec, or are they custom to gdspy?  What is the expected behavior here? I'm a little baffled that any memory is shared between the output of `pg.boolean` and its inputs, but I can look into it if that would be useful

closed 2025-03-12 by amccaugh

basnijholt, 2025-03-13:
Closed but not fixed, I guess?

I no longer use phidl, unfortunately. However, I was using it to store some additional data in the the polygons. Also they can in-principle be included in the GDS too. Although then it needs to be `dict[int, str]` only.