amccaugh/phidl #168: gf.path.smooth fails with near-collinear points

joamatab, 2023-01-06:
How can we have a smooth path that has near-collinear points?

```
import phidl.path as pp

p = pp.smooth(points=[(0, 0), (0, 1000), (1, 10000)])

c = p.extrude(width=1, layer=(1, 0))

# IndexError: index -1 is out of bounds for axis 0 with size 0
```

https://github.com/gdsfactory/gdsfactory/issues/1090

amccaugh, 2023-01-09:
Interesting.  It looks like this is an error in `pp.euler` when the angle is so small that there wouldn't be any points made.  I'll take a look

joamatab, 2023-01-09:
We fixed it on gdsfactory 

Feel free to take a look there

amccaugh, 2023-01-09:
I took a look, you may want to double-check the accurateness of the code.  I tried the same approach at first, but found that for various `p < 0.5` values errors still occurred

joamatab, 2023-01-10:
I see

let me know if you find a better fix

amccaugh, 2023-01-12:
I believe this is now fixed in https://github.com/amccaugh/phidl/commit/69178d8f492b7d98cad4dffba4f9fd98293a26d2

closed 2023-01-13 by amccaugh