amccaugh/phidl #173: Strange artifacts from path -> CrossSection -> extrude

basnijholt, 2023-01-17:
See the following code:
```python
import phidl.path as pp
from phidl.device_layout import CrossSection
import numpy as np
from phidl.quickplotter import quickplot as qp
import matplotlib.pyplot as plt

points = np.array(
    [
        [-6.6, -13.5],
        [-6.6, -16.5],
        [-38.3, -57.75],
        [-160.0, -59.75],
        [-240.0, -105.0],
        [-240.0, -107.0],
    ],
)

# Show the points
plt.scatter(*points.T)
plt.show()

path = pp.smooth(
    points=points,
    radius=0.005,
    corner_fun=pp.euler,
)
x = CrossSection()
x.add(width=6)
device = path.extrude(x)

qp(device)
```

Which tries to extrude the following line:
[image]

[image]

And results in:
[image]
with zoom ins:
[image]
[image]

I do not expect these artifacts in the corners. Is this my fault or is this a bug?

amccaugh, 2023-01-17:
Yes, those are a known artifact known as "cusps".  They're the result of trying to take low-radius turns with large width wires, see for instance the first image in https://en.wikipedia.org/wiki/Parallel_curve:

[image]

There are a variety of algorithms dedicated to "fixing" this, but after a lot of background research I decided to avoid them entirely for two reasons: (1) they're generally very slow, especially in Python, (2) it's not obvious what the proper behavior *should even be* -- depending on who you ask, they might want those corners to just be clipped (leaving a sharp interior angle):

[image]

or they may want them to be rounded as in a true offset operation:

[image]

basnijholt, 2023-01-18:
Thanks a lot for your reply. 

Could you perhaps point me to some of those algorithms that remove the cusps (one's you are aware of at least)?

joamatab, 2023-01-18:
Hi Bas,

I tried this in gdsfactory and seems to work  fine

```
import numpy as np
import matplotlib.pyplot as plt
import gdsfactory as gf

points = np.array(
    [
        [-6.6, -13.5],
        [-6.6, -16.5],
        [-38.3, -57.75],
        [-160.0, -59.75],
        [-240.0, -105.0],
        [-240.0, -107.0],
    ],
)

path = gf.path.smooth(
    points=points,
    radius=0.005,
)
x = gf.CrossSection(layer=(1, 0), width=2)
# x = gf.CrossSection(width=2)
c = path.extrude(x)
c.show()
```

amccaugh, 2023-01-18:
@joamatab that's very interesting -- do you know what's happening differently?  Typically the fixes I've seen for this problem are some kind of union/boolean operation that intrinsically is able to remove these kinds of self-intersections, but they come at a fairly high computational overhead

amccaugh, 2023-01-18:
@basnijholt I found that searching for the keywords `cusp`, `self-intersection`, `polyline` and `offset` ended up being the most helpful.  These stackexchange answers in particular were illuminating:

* https://math.stackexchange.com/questions/2208468/self-intersection-removal-in-offset-curves
* https://stackoverflow.com/questions/69935520/offsetting-a-polyline-in-one-direction
* https://math.stackexchange.com/a/302101

For instance, one of them points to this paper, which isn't terribly well-written but is helpful to understand the depth/variety of the problem and has some good references: http://www.cs.technion.ac.il/~gershon/papers/offset-compare.pdf .  Other work that seems a bit more robust (but also a little more impenetrable) is this: [An offset algorithm for polyline curves](https://www.researchgate.net/publication/222549414_An_offset_algorithm_for_polyline_curves)

amccaugh, 2023-01-18:
More to the point -- what's happening where there are cuts in the path that aren't present in the `path.extrude(x)` operation?

closed 2023-04-17 by amccaugh