amccaugh/phidl #178: Weird waveguide artifact

Ikechi12, 2023-04-07:
I noticed that the beginning of my waveguide is weird-shaped as shown in the screenshot.
I suspect it is because of how I scripted the vertical straight portion of the waveguide.  {I scripted the vertical straight portion by creating a 90 degrees arc with a zero radius and appended it to a straight waveguide with the desired vertical length. This is because starting a waveguide path with a straight waveguide only gives horizontal waveguide paths}.

Is there any way of avoiding this artifact? Is there another way of scripting vertical waveguide paths?
The script is shown below.

Ps: While we are here, is it possible to use a loop to generate the different waveguides instead of repeating it as I did?
 
[image]

```
from phidl import Device
from phidl import quickplot as qp # Rename "quickplot()" to the easier "qp()"
import phidl.geometry as pg # Rename "phidl.geometry" to the easier "pg"
from phidl import Path, CrossSection, Device
import phidl.path as pp
import numpy as np

# Create a blank device where polygons will be added
cb = Device('cutback')

P = Path()
P.append( pp.arc(radius = 0, angle = 90) )   # Circular arc
P.append( pp.straight(length = 1650) )          # Straight section
P.append( pp.arc(radius = 93.3, angle = -90) )  # Circular arc
P.append( pp.straight(length = 2500) )          # Straight section
P.append( pp.arc(radius = 93.3, angle = 90) )  # Circular arc
P.append( pp.straight(length = 1550) )          # Straight section

P1 = Path()
P1.append( pp.arc(radius = 0, angle = 90) )   # Circular arc
P1.append( pp.straight(length = 1620) )          # Straight section
P1.append( pp.arc(radius = 93.3, angle = -90) )  # Circular arc
P1.append( pp.straight(length = 2500) )          # Straight section
P1.append( pp.arc(radius = 93.3, angle = 90) )  # Circular arc
P1.append( pp.straight(length = 1580) )          # Straight section

P2 = Path()
P2.append( pp.arc(radius = 0, angle = 90) )   # Circular arc
P2.append( pp.straight(length = 1590) )          # Straight section
P2.append( pp.arc(radius = 93.3, angle = -90) )  # Circular arc
P2.append( pp.straight(length = 2500) )          # Straight section
P2.append( pp.arc(radius = 93.3, angle = 90) )  # Circular arc
P2.append( pp.straight(length = 1610) )          # Straight section

P3 = Path()
P3.append( pp.arc(radius = 0, angle = 90) )   # Circular arc
P3.append( pp.straight(length = 1560) )          # Straight section
P3.append( pp.arc(radius = 93.3, angle = -90) )  # Circular arc
P3.append( pp.straight(length = 2500) )          # Straight section
P3.append( pp.arc(radius = 93.3, angle = 90) )  # Circular arc
P3.append( pp.straight(length = 1640) )          # Straight section

P4 = Path()
P4.append( pp.arc(radius = 0, angle = 90) )   # Circular arc
P4.append( pp.straight(length = 1530) )          # Straight section
P4.append( pp.arc(radius = 93.3, angle = -90) )  # Circular arc
P4.append( pp.straight(length = 2500) )          # Straight section
P4.append( pp.arc(radius = 93.3, angle = 90) )  # Circular arc
P4.append( pp.straight(length = 1670) )          # Straight section

waveguide_device = cb << P.extrude(0.8, layer = 1)
waveguide_device1 = cb << P1.extrude(0.8, layer = 1)
waveguide_device2 = cb << P2.extrude(0.8, layer = 1)
waveguide_device3 = cb << P3.extrude(0.8, layer = 1)
waveguide_device4 = cb << P4.extrude(0.8, layer = 1)
cb.distribute(direction = 'x', spacing = -2614.6)

qp(cb)

cb.write_gds('Cutback.gds')
```

amccaugh, 2023-04-17:
Yes, you're absolutely right for the reason why it's causing this strange issue -- it's because pp.arc(radius=0) is a degenerate condition, and it's producing errors.  I may have to add a user warning to avoid people from that. 

Fortunately, there's a really easy fix.  Just rotate() your Path, which will allow you to orient it any direction you like, even if the Path doesn't have any segments/points in it yet:

```python
P = Path().rotate(90)
P.append( pp.straight(length = 1650) )          # Straight section
P.append( pp.arc(radius = 93.3, angle = -90) )  # Circular arc
P.append( pp.straight(length = 2500) )          # Straight section
P.append( pp.arc(radius = 93.3, angle = 90) )  # Circular arc
P.append( pp.straight(length = 1550) )          # Straight section
```

Ikechi12, 2023-04-18:
Thank you @amccaugh

closed 2023-04-18 by amccaugh