amccaugh/phidl #179: Suggest Change to Transition

TravisAutry, 2023-05-26:
Hey,

I just found a bug.  On the transition function it reads

`     offset_fun = _sinusoidal_transition(offset1, offset2)

            if width_type == "sine":
                width_fun = _sinusoidal_transition(width1, width2)
            elif width_type == "linear":
                width_fun = _linear_transition(width1, width2)
            else:
                raise ValueError(
                    "[PHIDL] transition() width_type "
                    + "argument must be one of {'sine','linear'}"
                )`

I think it would be better if 

``     

            if width_type == "sine":
                width_fun = _sinusoidal_transition(width1, width2)
                offset_fun = _sinusoidal_transition(offset1, offset2)
            elif width_type == "linear":
                width_fun = _linear_transition(width1, width2)
                offset_fun = _linear_transition(offset1, offset2)
            else:
                raise ValueError(
                    "[PHIDL] transition() width_type "
                    + "argument must be one of {'sine','linear'}"
                )``

amccaugh, 2023-05-31:
Interesting, any chance you can try out this change on https://github.com/amccaugh/phidl/issues/177 ?  I'm curious if it might actually solve that problem.

Aside from that, can you describe why it might be better?  I believe I avoided allowing this because it caused some strange artifacts, but that was a while ago.

amccaugh, 2023-06-06:
Do you have a minimum-working example you can share?

loganbvh, 2024-01-03:
> Do you have a minimum-working example you can share?

I came across this same issue with the code below:

```python
width = 500
gap = width * 0.5
offset = (width + gap) / 2

X1 = CrossSection()
X1.add(width=width, offset=0, layer=1, name="center", ports=("in", "out"))
X1.add(width=gap, offset=-offset, layer=0, name="left")
X1.add(width=gap, offset=offset, layer=0, name="right")

width = 100
gap = width * 0.5
offset = (width + gap) / 2

X2 = CrossSection()
X2.add(width=width, offset=0, layer=1, name="center", ports=("in", "out"))
X2.add(width=gap, offset=-offset, layer=0, name="left")
X2.add(width=gap, offset=offset, layer=0, name="right")

X = pp.transition(X1, X2, width_type="linear")
P = pp.straight(length=500)
D = P.extrude(X)

qp(D)
```

[image]

It seems like the fix proposed [above](https://github.com/amccaugh/phidl/issues/179#issue-1728154645) would fix this.

loganbvh, 2024-01-10:
Makes sense, thanks for the explanation!

closed 2024-01-15 by amccaugh