amccaugh/phidl #182: Port not working on extruded waveguide

Ikechi12, 2023-06-30:
Hello. I have been unable to add ports to my extruded waveguides. Why is this so? It appears that the ports feature doesn't work on extruded waveguides. My goal is to add ports on each waveguide. Please what should I do differently? The script is shown below

```
from phidl import Device
from phidl import quickplot as qp # Rename "quickplot()" to the easier "qp()"
import phidl.geometry as pg # Rename "phidl.geometry" to the easier "pg"
from phidl import Path, CrossSection, Device
import phidl.path as pp
import numpy as np

# Create a blank device where polygons will be added
cutback = Device('cutback')
wg = Device()

wg_width = 0.31
wg_bend_rad = 100
spacing = 3
i = 0
pair_separation = 70

A = CrossSection()
A.add(width=wg_width)

# Section 1
length1 = [1650, 1620, 1590, 1560, 1530]
length2 = [1550, 1580, 1610, 1640, 1670]
for l1, l2 in zip(length1, length2):
    # Create waveguide
    path1 = Path().rotate(90) #save waveguide in path1
    path1.append(pp.straight(length=l1))          # Straight section
    path1.append(pp.arc(radius=wg_bend_rad, angle=-90))  # Circular arc
    path1.append(pp.straight(length=1600))          # Straight section
    path1.append(pp.arc(radius=wg_bend_rad, angle=90))  # Circular arc
    path1.append(pp.straight(length=l2))         # Straight section
    # Extrude waveguide 
    P = path1.extrude(A)
    P.add_port(name=None, midpoint=(0, 0), width=100, orientation=-90, port=None) #add port
    # Position waveguide
    if i > 0:
        P = P.movex(origin=0, destination = (i*((0.31)+pair_separation)))
    i = i+1  
    wg = wg.add(P)
qp(wg)
```

amccaugh, 2023-07-01:
This is actually a pretty simple error -- the only part that needs changing is changing the line

```
    wg = wg.add(P)
```
to
```
    wg.add_ref(P)
```

That's because phidl Devices don't use the .add() function for Devices (I should probably put a warning in here, it's a little confusing) -- instead, you're supposed to use specific functions like `add_port()` or `add_ref()` or `add_polygon()`. So when you call `wg.add()`, it's actually calling gdspy's `Cell.add()` function which doesn't support ports, hence why they disappear.

Also, I'll mention there's a much easier way to add ports to a waveguide -- define htem in your crossection with the `ports = ...` argument, for instance:

```python
from phidl import Device
from phidl import quickplot as qp # Rename "quickplot()" to the easier "qp()"
import phidl.geometry as pg # Rename "phidl.geometry" to the easier "pg"
from phidl import Path, CrossSection, Device
import phidl.path as pp
import numpy as np

# Create a blank device where polygons will be added
cutback = Device('cutback')
wg = Device()

wg_width = 0.31
wg_bend_rad = 100
spacing = 3
i = 0
pair_separation = 70

A = CrossSection()
A.add(width=wg_width, ports = (1,2)) # MODIFIED

# Section 1
length1 = [1650, 1620, 1590, 1560, 1530]
length2 = [1550, 1580, 1610, 1640, 1670]
for l1, l2 in zip(length1, length2):
    # Create waveguide
    path1 = Path().rotate(90) #save waveguide in path1
    path1.append(pp.straight(length=l1))          # Straight section
    path1.append(pp.arc(radius=wg_bend_rad, angle=-90))  # Circular arc
    path1.append(pp.straight(length=1600))          # Straight section
    path1.append(pp.arc(radius=wg_bend_rad, angle=90))  # Circular arc
    path1.append(pp.straight(length=l2))         # Straight section
    # Extrude waveguide 
    P = path1.extrude(A)
    # DELETED - P.add_port(name=None, midpoint=(0, 0), width=100, orientation=-90, port=None) #add port
    # Position waveguide
    if i > 0:
        P = P.movex(origin=0, destination = (i*((0.31)+pair_separation)))
    i = i+1  
    wg.add_ref(P) # MODIFIED
qp(wg)
```

[image]

Ikechi12, 2023-07-02:
Oh! Yeah. I see. Thanks for the clarification and insight @amccaugh. You have been so helpful.

Ikechi12, 2023-07-02:
I am also finding it difficult to connect the ports using your updated script. Is there something that I am fundamentally missing in my understanding of Phidl?

So I created a taper function

```
from phidl import Device
from phidl import quickplot as qp # Rename "quickplot()" to the easier "qp()"
import phidl.geometry as pg # Rename "phidl.geometry" to the easier "pg"
from phidl import Path, CrossSection, Device
import phidl.path as pp
import numpy as np

def taperfnc (wg_width):
    
    taper = Device()
    # D1 = Device()
    # wg_width = 0.31
    taper_length = 200
    small_wg_width = 0.118  #narrow end of taper
    
    # taper_left = pg.taper(length = 200, width1 = wg_width, width2 = 0.118, port = None, layer = 0)
    # taper_left.rotate(-90)
    # qp(taper_left)
    
    # Create the waveguide path
    P = Path().rotate(90)
    P.append(pp.straight(taper_length))
    tap = P.extrude([small_wg_width,wg_width])
    taper.add(tap)  #add the taper to taper device
    
    # Create ruler
    
    D1 = pg.litho_ruler(  #left ruler
        height = 10 ,
        width = 1,
        spacing = 20,
        scale = [2,1,1,1,1,2,1,1,1,1,2],
        num_marks = 11,
        layer = 0,
        )
    D2 = pg.litho_ruler( #right ruler
        height = 10 ,
        width = 1,
        spacing = 20,
        scale = [2,1,1,1,1,2,1,1,1,1,2],
        num_marks = 11,
        layer = 0,
        )
    
    # Position rulers
    D1.rotate(90).movex(origin=0,destination=(-12.5))
    D2.rotate(90).mirror().movex(origin=0,destination=(12.5))
    '''
    The above code D2.rotate(90).mirror().movex(origin=0,destination=(10)) 
    is equivalent to
    D2.rotate(90)
    D2.mirror()
    D2.movex(origin=0,destination=(10))
    '''
    # Add rectangle
    rect = pg.rectangle(size = (5, 5))
    rect.move(origin=(0,0), destination=(-2.5,-5))
    taper.add(rect) #add rectangle to taper device
    #add rulers to taper device
    taper.add(D1).add(D2)
    
    taper.add_port(name='1', midpoint=(0, taper_length), width=1, orientation=90, port=None)
    return taper
```

and I added  the function to the body of the main code 

```
from phidl import Device
from phidl import quickplot as qp # Rename "quickplot()" to the easier "qp()"
import phidl.geometry as pg # Rename "phidl.geometry" to the easier "pg"
from phidl import Path, CrossSection, Device
import phidl.path as pp
import numpy as np
from taper_function import taperfnc

# Create a blank device where polygons will be added
cutback = Device('cutback')
wg = Device()

wg_width = 0.31
wg_bend_rad = 100
spacing = 3
i = 0
pair_separation = 70

A = CrossSection()
A.add(width=wg_width, ports = (1,2)) # MODIFIED

taper1 = taperfnc(wg_width) #down taper
taper2 = taper1.rotate(180) #up taper

# Section 1
length1 = [1650, 1620, 1590, 1560, 1530]
length2 = [1550, 1580, 1610, 1640, 1670]
for l1, l2 in zip(length1, length2):
    # Create waveguide
    path1 = Path().rotate(90) #save waveguide in path1
    path1.append(pp.straight(length=l1))          # Straight section
    path1.append(pp.arc(radius=wg_bend_rad, angle=-90))  # Circular arc
    path1.append(pp.straight(length=1600))          # Straight section
    path1.append(pp.arc(radius=wg_bend_rad, angle=90))  # Circular arc
    path1.append(pp.straight(length=l2))         # Straight section
    # Extrude waveguide 
    P = path1.extrude(A)
    # REINTRODUCED - P.add_port(name=None, midpoint=(0, 0), width=100, orientation=-90, port=None) #add port
    # Position waveguide
    if i > 0:
        P = P.movex(origin=0, destination = (i*((0.31)+pair_separation)))
    i = i+1  
    #t1 = wg.add_ref(taper1)
    t2 = wg.add_ref(taper2)
    wg.add_ref(P) # MODIFIED
    #t1.connect(port = 1, destination = wg.ports['2'])
    t2.connect(port = 1, destination = wg.ports['1'])
qp(wg)
```

I have the following error

```
runfile('C:/Users/...../Python/untitled4.py', wdir='C:/Users/..../Python')
Reloaded modules: taper_function
Traceback (most recent call last):

  File "C:\Users\......\untitled4.py", line 54, in <module>
    t2.connect(port = 1, destination = wg.ports['1'])

KeyError: '1'
```
However, when I include the port as I did previously:

```
from phidl import Device
from phidl import quickplot as qp # Rename "quickplot()" to the easier "qp()"
import phidl.geometry as pg # Rename "phidl.geometry" to the easier "pg"
from phidl import Path, CrossSection, Device
import phidl.path as pp
import numpy as np
from taper_function import taperfnc

# Create a blank device where polygons will be added
cutback = Device('cutback')
wg = Device()

wg_width = 0.31
wg_bend_rad = 100
spacing = 3
i = 0
pair_separation = 70

A = CrossSection()
A.add(width=wg_width) # UNMODIFIED

taper1 = taperfnc(wg_width) #down taper
taper2 = taper1.rotate(180) #up taper

# Section 1
length1 = [1650, 1620, 1590, 1560, 1530]
length2 = [1550, 1580, 1610, 1640, 1670]
for l1, l2 in zip(length1, length2):
    # Create waveguide
    path1 = Path().rotate(90) #save waveguide in path1
    path1.append(pp.straight(length=l1))          # Straight section
    path1.append(pp.arc(radius=wg_bend_rad, angle=-90))  # Circular arc
    path1.append(pp.straight(length=1600))          # Straight section
    path1.append(pp.arc(radius=wg_bend_rad, angle=90))  # Circular arc
    path1.append(pp.straight(length=l2))         # Straight section
    # Extrude waveguide 
    P = path1.extrude(A)
    P.add_port(name='1', midpoint=(0, 0), width=100, orientation=-90, port=None) #add port
    # Position waveguide
    if i > 0:
        P = P.movex(origin=0, destination = (i*((0.31)+pair_separation)))
    i = i+1  
    #t1 = wg.add_ref(taper1)
    t2 = wg.add_ref(taper2)
    wg.add_ref(P) # MODIFIED
    # t1.connect(port = 1, destination = wg.ports['2'])
    t2.connect(port = '1', destination = P.ports['1'])
qp(wg)
```
It gives me the desired feature. **Why is this so?**
[image]
[image]

amccaugh, 2023-07-05:
Take a look at the error closely -- you'll see you're mixing up the names of the port when trying to connect.  One of them has a port named with the **string** `'1'`, but the port you're trying to connect to has a port named with the  **integer** `1`

closed 2023-07-12 by amccaugh