amccaugh/phidl #183: Use of deprecated `np.bool` raises error in `phidl.geometry.fill_rectangle`

MechanicalPenguin225, 2023-07-11:
Hello, I've been using phidl for a bit now and wanted to create a ground plane for waveguides using `phidl.geometry.fill_rectangle`. However, I get a numpy error - apparently, since numpy 1.20.0, `np.bool` as a datatype is deprecated. Here's a quick program recreating the issue : 

```
from phidl import Device, LayerSet
from phidl import quickplot as qp

import phidl.geometry as pg
import matplotlib.pyplot as plt

ls = LayerSet()
ls.add_layer(name = 'optical', 
            gds_layer = 1,
            gds_datatype = 0,
            description = 'layer for laser lithography',
            color = 'lightcoral')

ls.add_layer(name = 'ground_avoidance',
                    gds_layer = 10,
                    gds_datatype = 0,
                    description = 'layer for ground avoidance',
                    color = 'lightyellow')

chip = Device()

box = chip << pg.rectangle()

gp = chip << pg.fill_rectangle(D = chip,
                    fill_size= (10,10),
                    avoid_layers=[ls['ground_avoidance']],
                    margin= 10,
                    fill_layers= [ls['optical']],
                    fill_densities= 0.25,
                    fill_inverted=False)

qp(chip)
plt.show()```

I am running numpy 1.24.3 in a conda environment.  When running the Python script above, I get the following error : 

```
C:\Users\XXXX\AppData\Local\miniconda3\envs\stage-m2\Lib\site-packages\phidl\geometry.py:3776: FutureWarning: In the future `np.bool` will be defined as the corresponding NumPy scalar.
  raster = np.zeros((ysize, xsize), dtype=np.bool)
Traceback (most recent call last):
  File "C:\Users\XXXX\Documents\Stage M2\chip-design\v0\mvbr.py", line 25, in <module>
    gp = chip << pg.fill_rectangle(D = chip,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\XXXX\AppData\Local\miniconda3\envs\stage-m2\Lib\site-packages\phidl\geometry.py", line 3981, in fill_rectangle
    raster = _rasterize_polygons(
             ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\XXXX\AppData\Local\miniconda3\envs\stage-m2\Lib\site-packages\phidl\geometry.py", line 3776, in _rasterize_polygons
    raster = np.zeros((ysize, xsize), dtype=np.bool)
                                            ^^^^^^^
  File "C:\Users\XXXX\AppData\Local\miniconda3\envs\stage-m2\Lib\site-packages\numpy\__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'bool'.
`np.bool` was a deprecated alias for the builtin `bool`. To avoid this error in existing code, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations. Did you mean: 'bool_'?
```

- Lou

amccaugh, 2023-07-12:
Thanks for the bug report, I'll go ahead and fix that.  I'm surprised anyone found this rather hidden, and admittedly poorly-documented functionality!

amccaugh, 2023-07-24:
Fixed in #184 and added new documentation to explain its usage better

closed 2023-07-24 by amccaugh