amccaugh/phidl #185: Roughness of bends, curves and rings edges

Ikechi12, 2023-08-29:
I have noticed that the edges of bends and curves generated using phidl are not smooth. This affected the fabricated devices as well.
I have tried a couple of methods:
1. using paths and increasing the number of points
2. using ring and curve polygons and increasing the angle_resolution.
However, the issue persists.

Below is an image of a section of device I designed using Phidl:
[image]

Upon zooming, you will see that the curve is not so smooth, 
[image]
[image]

After fabrication, the images from the SEM show this roughness:
[image]
[image]

So how can I get the smoothest possible curves and rings using Phidl?

amccaugh, 2023-09-13:
Hi @Ikechi12 , apologies for the delay in response.  I believe what you want here is *increased* `angle_resolution`, not decreased -- the `angle_resolution` sets the number of degrees per point, so if `angle_resolution = 1`, a full ring will consist of 360 points.  If `angle_resolution = 0.5`, a full ring will consist of 720 points.  For example:

```python
import phidl.geometry as pg

D =  pg.ring(radius = 5, width = 0.5, angle_resolution = 1, layer = 0)
D << pg.ring(radius = 5, width = 0.5, angle_resolution = 0.2, layer = 1)
D.write_gds('rings.gds')
```
`angle_resolution = 1`
[image]

`angle_resolution = 0.2`
[image]

amccaugh, 2023-09-13:
Also, note that *all* structures in GDSII are polygons -- there are no representations of perfect arcs/circles.  By adding a lot of points, however, you can generate curves that should have much higher resolution than what is achievable even with e-beam, so they should look as good as practically possible

Ikechi12, 2023-09-18:
Thank you @amccaugh

closed 2023-09-19 by amccaugh