amccaugh/phidl #188: Path.append() offset and direction error

Fabdef, 2023-12-11:
I am using VScode on Windows 11. 
I am new to using Phidl and I just wanted to make a very basic test (corresponding to want I need to do with it in a more complex project): 

pts1 = [(1, 0), (1, 5), (5, 5)]
P = Path()
P.append( pts1 )
qp(P)

Unfortunately, the path displayed by qp does not start at (1,0) but at (0,0), and it does not end at (5,5) but at (5,-4).
Am I doing something wrong? 

When doing Path(pts1) instead of P.append, we get the correct result. 
qp(Path(pts1))

What is going on with Path.append? Am I doing something wrong or is it a bug?

[image]
[image]

amccaugh, 2023-12-11:
You're correct that `Path(pts1)` is the right way to initialize a set of points.  What you're seeing is that when you call `Path()`, it initializes a new path that has the initial point `(0,0)` and is pointing with a zero-degree orientation (in the +x direction).  When you `append` to that path, it's treating the new list of points as a second path that you want to "snap" to the first.  So it draws your second path (as shown in your bottom figure) and then "snaps" it to `(0,0)` pointing right

Fabdef, 2023-12-12:
Thank you for the explanation. Would you consider it as a "feature"? If so I would strongly recommend to have a mention of that behavior added to the doc because the way I understood the doc is that both ways were equivalent.
In the doc (Paths and waveguides section), the first sentences describing the command Path say: 
_Creating a path device is simple:
 - Create a blank Path
 - Append points to the Path either using the built-in functions (arc(), straight(), euler(), etc) or by providing your own lists of points_
 
 Which is exactly what I did: I created a blank path with Path(), then I provided it my own list of points, which creates a path that is different from the list of points I provide. 
 So there is clearly an inconsistency somewhere, either in the doc or in Path() behavior.

amccaugh, 2023-12-14:
That's a very reasonable point, I'm always in favor of clarifying documentation, especially targeting newer users. Do you have any suggestions on how to word it maybe?  How about:

> Creating a path device is simple:
> 
>     Use a list of points to create your own custom Path
>     and/or:
>     Snap together multiple Paths with the built-in functions (arc(), straight(), euler(), etc)

amccaugh, 2024-04-11:
Addressed in a4ee9288f4e3b36df0b7885b54b98796634bedaa

closed 2024-04-11 by amccaugh