amccaugh/phidl #191: Moving objects along a circle

paniash, 2024-02-22:
Hi,

I just came across this wonderful package and it's been just great! I wanted to ask if there is a method to move objects along a circle? i.e. by providing a radius, center and angle, onto which the object can be placed?

Many thanks!

amccaugh, 2024-04-09:
Thanks for the nice feedback.  Can you elaborate a little?  You can always calculate the position along a circle using some basic trigonometry, or are you asking something else?

paniash, 2024-04-09:
@amccaugh I was wondering if there is a method within the package itself to assign a center and angle. But yes, since I last posted I was able to implement my task with basic trigonometry. :)

closed 2024-04-09 by paniash

amccaugh, 2024-04-09:
Glad to hear it!