amccaugh/phidl #200: This is a great project, but I have a few questions about the pp.transtion function

fuckingphoto, 2024-04-22:
Here is my code:

from phidl import path,CrossSection,Device
import phidl.path as pp
from phidl import set_quickplot_options
from phidl import quickplot as qp
set_quickplot_options(new_window=True)
X1=CrossSection()
X1.add(width=1.2,offset=0,layer=2,name='wg',ports=('1','11'))
X1.add(width=2.2,offset=0,layer=3,name='wg1',ports=('2','22'))
X1.add(width=1.1,offset=3,layer=1,name='wg2')

X2=CrossSection()
X2.add(width=1,offset=0,layer=2,name='wg',ports=('in2','out2'))
X2.add(width=3.5,offset=0,layer=3,name='wg1')
X2.add(width=3,offset=5,layer=1,name='wg2')

P=pp.Path()
P1=pp.straight(length=5)
P2=pp.straight(length=15)
WG1=P1.extrude(X1)
WG2=P2.extrude(X2)

D=Device()
wg1=D<<WG1
wg2=D<<WG2
wg2.movex(7.5)

qp(D)

Xtrans=pp.transition(cross_section1=X1,cross_section2=X2,width_type='linear')
P3=pp.straight(length=10)
WGtrans=P2.extrude(Xtrans)
qp(WGtrans)
input()
As you can see I have a couple of questions when I run this piece of code, the first one I don't quite understand is why do I need to add the name item to crosssection.I think it seems to be a more troublesome way.The second problem is that when I use the”linear" code, the transition area of the purple square is still sine transition. As you can see:
[image]
In the end, it seems that the naming logic of the port is strange, maybe the port of the Xtransition device is cross_section1 's out1 instead of in2?
Thank you very much for your reading. If you are willing to answer my questions, I will express my most sincere thanks to you.

amccaugh, 2024-04-22:
* For needing `name`, the reason is that the code needs to know which parts of  `X2` connects to which parts of `X1`.  For instance, in the above code, what if you wanted the transition to connect the purple waveguide to the yellow part?
* For the width, you are partially correct -- the **width** is tapering linearly, but the **offset** is still a sinuisoidal

Timemakesmegrow, 2024-04-23:
Thank you very much for your explanation. I understand the reason for the first name question, But I don't understand the second question because I didn't set offset to sinusoidal

amccaugh, 2024-04-23:
This might provide some better context: https://github.com/amccaugh/phidl/issues/179

Timemakesmegrow, 2024-04-24:
I understand now. Can this problem be solved now? If not, I wonder if It can create an optional point in Xtransition to help create path orientation and trapezoid orientation corrections? If not, thank you very much for your answers, and I look forward to you updating more waveguide code examples.Your project is one of the best gds projects I've ever seen

amccaugh, 2024-04-24:
If you can think of a way to solve it please let me know!

closed 2024-04-24 by amccaugh