amccaugh/phidl #203: Duplicate boundary points extruding an arc

EliDoris, 2024-05-29:
I defined a path via `pp.arc()` using a cross section such that the width, offset, and radius resulted in a perfect section of a circle. For example:
`radius = 10`
`width = 20`
`offset = 0`

At the center of the circle this resulted in degenerate boundary points which caused problems later on in some DRC software

[image]

amccaugh, 2024-05-29:
That's pretty funny!  I'm not sure what to do about it exactly though, it's working as designed.  Did you have some functionality in mind?

EliDoris, 2024-05-29:
I think it might be nice to have a function that goes through polygons and gets rid of duplicate points. This could serve as sort of a local or global "heal" kind of thing that would fix this issue.

I was able to get around this in my own design using a different type of "heal" function that does a positive and negative offset of the same amount on that particular layer. Since this does some geometry recalculation it got rid of the duplicate point.

We can close the issue, just ideas for future improvements at this point since it's a minor edge case

amccaugh, 2024-05-29:
You're in luck!  This already exists:  https://phidl.readthedocs.io/en/latest/tutorials/waveguides.html#Simplifying-/-reducing-point-usage  . This can be applied directly to devices too.  For instance: 

```python
from phidl import Path
import phidl.path as pp
from phidl import quickplot as qp

P = pp.arc(radius = 10, angle = 90)
D = P.extrude(20)
D.simplify(0.001)
qp(D)
```

EliDoris, 2024-05-30:
Excellent, I'll check this out as a way to fix things up

closed 2024-05-30 by EliDoris