amccaugh/phidl #207: extrude() failing for certain paths

pvasired, 2024-07-24:
I have a somewhat strange path

```
a_star_path = np.array([[  3870.,   -645.],
       [  4515.,   -645.],
       [  4515.,      0.],
       [  4515.,    645.],
       [  4515.,   1290.],
       [  4515.,   1935.],
       [  4515.,   2580.],
       [  4515.,   3225.],
       [  4515.,   3870.],
       [  3870.,   3870.],
       [  3870.,   4515.],
       [  3870.,   5160.],
       [  3870.,   5805.],
       [  3225.,   5805.],
       [  2580.,   5805.],
       [  2580.,   6450.],
       [  1935.,   6450.],
       [  1290.,   6450.],
       [   645.,   6450.],
       [     0.,   6450.],
       [     0.,   6450.],
       [  -645.,   6450.],
       [ -1290.,   6450.],
       [ -1935.,   6450.],
       [ -2580.,   6450.],
       [ -2580.,   5805.],
       [ -3225.,   5805.],
       [ -3870.,   5805.],
       [ -4515.,   5805.],
       [ -5160.,   5805.],
       [ -5805.,   5805.],
       [ -6450.,   5805.],
       [ -7095.,   5805.],
       [ -7740.,   5805.],
       [ -8385.,   5805.],
       [ -8385.,   6450.],
       [ -8385.,   7095.],
       [ -8385.,   7740.],
       [ -9030.,   7740.],
       [ -9675.,   7740.],
       [-10320.,   7740.],
       [-10320.,   8385.]])
```

I want to extrude this path (ideally with a CrossSection). However, when I do a simple

```
P = Path(a_star_path)
path = P.extrude(trace_width, layer=layer_number)
D.add_ref(path)
```

The path becomes undefined. The `path.xmin` and `path.xmax` become infinite and lead to nans which eventually culminate in this warning:
```
...\Lib\site-packages\phidl\device_layout.py:3060: RuntimeWarning: divide by zero encountered in divide
  offset_distance = offset_distance / np.sin(dtheta_int / 2)
...\Lib\site-packages\gdspy\polygon.py:232: RuntimeWarning: invalid value encountered in cast
  xy = numpy.round(self.polygons[ii] * multiplier).astype(">i4")
  ```

When I then try to do a `D.write_gds(filename, cellname=top_cell_name)`, the extruded path does not show up. Interestingly, this code works totally fine for certain paths but then breaks for other paths, even though the paths all seem valid to me. Any ideas here?

amccaugh, 2024-07-25:
Your path repeats the ` [     0.,   6450.],` point twice, and as a result the path extrapolation algorithm is dividing by zero.  You should just be able to remove the redundant one and it works just fine:
     
```
from phidl import Device, quickplot as qp, Path
import phidl.geometry as pg

a_star_path = np.array([[  3870.,   -645.],
       [  4515.,   -645.],
       [  4515.,      0.],
       [  4515.,    645.],
       [  4515.,   1290.],
       [  4515.,   1935.],
       [  4515.,   2580.],
       [  4515.,   3225.],
       [  4515.,   3870.],
       [  3870.,   3870.],
       [  3870.,   4515.],
       [  3870.,   5160.],
       [  3870.,   5805.],
       [  3225.,   5805.],
       [  2580.,   5805.],
       [  2580.,   6450.],
       [  1935.,   6450.],
       [  1290.,   6450.],
       [   645.,   6450.],
       [     0.,   6450.],
       [  -645.,   6450.],
       [ -1290.,   6450.],
       [ -1935.,   6450.],
       [ -2580.,   6450.],
       [ -2580.,   5805.],
       [ -3225.,   5805.],
       [ -3870.,   5805.],
       [ -4515.,   5805.],
       [ -5160.,   5805.],
       [ -5805.,   5805.],
       [ -6450.,   5805.],
       [ -7095.,   5805.],
       [ -7740.,   5805.],
       [ -8385.,   5805.],
       [ -8385.,   6450.],
       [ -8385.,   7095.],
       [ -8385.,   7740.],
       [ -9030.,   7740.],
       [ -9675.,   7740.],
       [-10320.,   7740.],
       [-10320.,   8385.]])

D = Device()
P = Path(a_star_path)
path = P.extrude(1, layer=1)
D.add_ref(path)
qp(D)
```

pvasired, 2024-07-25:
Ah thanks so much, sorry about the simple mistake. Good catch!

closed 2024-07-25 by pvasired