amccaugh/phidl #208: Cross-section transition

alpasla, 2024-10-04:
The *linear* transition between cross-sections does not work well: the offset sections still taper with a sinusoidal profile. This is evident in the transition method defintion in phidl.path. However forcing the offset sections to be follow linear profiles looks bad too. The transition with the sine profile does work well.

Here is a minimum working example:
```
import phidl
import phidl.path as pp

readout_layer = (1,0)

readout_CPW_strip_w = 2
readout_CPW_slot_w  = 3
readout_CPW_w       = readout_CPW_strip_w + 2*readout_CPW_slot_w

bondpad_CPW_strip_w = 20
bondpad_CPW_slot_w  = 30
bondpad_CPW_w         = bondpad_CPW_strip_w + 2*bondpad_CPW_slot_w

readout_xsection      = phidl.CrossSection()
readout_xsection.add(width=readout_CPW_strip_w, offset=0, layer=readout_layer, name='strip')
readout_xsection.add(width=readout_CPW_slot_w, offset=+(readout_CPW_strip_w/2+readout_CPW_slot_w/2), layer=readout_layer, name='slot1')
readout_xsection.add(width=readout_CPW_slot_w, offset=-(readout_CPW_strip_w/2+readout_CPW_slot_w/2), layer=readout_layer, name='slot2')

bondpad_xsection      = phidl.CrossSection()
bondpad_xsection.add(width=bondpad_CPW_strip_w, offset=0, layer=readout_layer, name='strip')
bondpad_xsection.add(width=bondpad_CPW_slot_w, offset=+(bondpad_CPW_strip_w/2+bondpad_CPW_slot_w/2), layer=readout_layer, name='slot1')
bondpad_xsection.add(width=bondpad_CPW_slot_w, offset=-(bondpad_CPW_strip_w/2+bondpad_CPW_slot_w/2), layer=readout_layer, name='slot2')

bondpad_taper_xsection = pp.transition(cross_section1 = bondpad_xsection,
                                       cross_section2 = readout_xsection,
                                       width_type = 'linear')    

bondpad_taper_l = 100
bondpad_taper_path = pp.straight(length = bondpad_taper_l)
bondpad_taper          = bondpad_taper_path.extrude(bondpad_taper_xsection)

phidl.quickplot(bondpad_taper)

```

MechanicalPenguin225, 2025-02-04:
Just stumbled on this too. Looks like it might happen when a section is offset from the crossection center ?

amccaugh, 2025-02-18:
Interesting, thanks for the issue and the PR -- give me a bit of time to take a look!

Out of curiousity, would you say it fixes the issues described in https://github.com/amccaugh/phidl/issues/179 ?

MechanicalPenguin225, 2025-02-18:
Argh, I didn't see that issue... forgot to check out closed issues before doing the PR. I may or may not have done the exact same thing, and didn't notice the problem this approach has because I tried it on relatively long tapers. My bad !

closed 2025-03-12 by amccaugh